adafruit/Adafruit-GFX-Library #410: getTextBounds not working in unicode

Crapy, 2022-12-10:
So I am using thr standard Arduino library as well as gfx to display text in an e ink display, I have used a custom arial font and cannot figure out how to align a text to the right so the start of the text is all on the same exact horizontal line, since I am using an RTL language it will look better that way

i tried to use getTextBounds but it doesn't appear to correctly calculate the width

```int16_t  x1, y1;

uint16_t w, h;

display.getTextBounds(string, 0, 0, &x1, &y1, &w, &h);

uint16_t fullx = x - w;

display.setCursor(fullx, 10);
```

It does work when using a string that only contains English but not unicode language (Hebrew)

Any idea what am I doing wrong? Or is it a problem with char  byte calculation?

PaintYourDragon, 2022-12-10:
Unfortunately only 7- to 8-bit charsets are supported, as GFX needs to run on small devices going back to AVR (e.g. Arduino Uno).

There are some good alternatives if you’re using a 32-bit microcontroller, have a look at LovyanGFX and LVGL. The former is based somewhat on Adafruit_GFX so it might be an easier transition. Latter can produce really polished UIs but you’d be starting fresh.
https://github.com/lovyan03/LovyanGFX
https://lvgl.io

mikeysklar, 2026-11-12:
GFX reads one byte per character, so UTF-8 Hebrew (2 bytes per letter) can't be measured or drawn. An 8-bit workaround that runs on any board:

1. Convert the Hebrew letters U+05D0 to U+05EA: `fontconvert FreeSerif.ttf 28 1488 1514 > FreeSerifHebrew28pt.h`
2. In the header, change the last line's `0x5D0, 0x5EA` to `0x80, 0x9A`. Byte `0x80 + n` is now letter n (א is `\x80`, ת is `\x9A`).
3. Store each word in visual order, last letter first. The font has no space, so move the cursor between words:

```cpp
const char *shalom = "\x8D\x85\x8C\x99"; // ם ו ל ש

display.setFont(&FreeSerifHebrew28pt);
display.getTextBounds(shalom, 0, 0, &x1, &y1, &w, &h);
display.setCursor(rightEdge - w - x1, y);
display.print(shalom);
```

`getTextBounds()` then measures correctly. Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing:

[image]

#414 adds real UTF-8 glyph support. Closing this one, please reopen if the remap doesn't work for you.

closed 2026-11-12 by mikeysklar