adafruit/Adafruit-GFX-Library #415: GFX lib with font, canvas, and drawBitmap() runs slow

norbinz, 2023-02-13:
I breadboarded an Adafruit 3677 (ItsyBitsy 32u4 5V) with an Adafruit 4383 (240x135 LCD ST7789).
I'm seeing very slow run-time of the GFX library's drawBitmap().
In the Adafruit forum I asked what's the reason for the slowness, and does the library offer a faster alternative?
I shared a demonstration sketch that includes the slow drawBitmap(), and a faster code that I whipped up.
A support (?) fellow suggested that I ask on github.
Here's the forum discussion:
https://forums.adafruit.com/viewtopic.php?p=960409

I understand that the GFX lib has a Prime Directive, but I'm not sure if that applies here.

norbinz, 2023-02-14:
Oops, broken URL. Trying again:
https://forums.adafruit.com/viewtopic.php?p=960155

jsseab, 2024-08-14:
I noticed this problem too. I have an application that draws a canvas that is 141 x 19 pixels, and this takes around 183 ms with an ATMega644 running at 10 MHz (it's a custom Arduino-compatible board).

I looked at the implementation and realized one big factor is how each pixel write is done with an address window that is 1x1 pixels. This is obviously not a very good idea. A much better implementation is to setup the whole window first, then just write all pixels in sequence.

I changed the code and got about 23 ms execution time (8 times faster!). I will try and submit a pull request for this, but I am a little unsure of exactly how to implement the change due to the class hierarchy.

jsseab, 2025-07-11:
Any reason why my PR is not accepted yet? Is there anything else that is needed from my side?

ladyada, 2025-07-13:
we don't have a full time maintainer for this library