adafruit/Adafruit-GFX-Library #447: fontconvert compilation error on MacOS Sonoma M1

mitch-spec, 2024-01-18:
I am trying to compile fontconvert on MacOS Sonoma M1.
freetype 2.13.2 is already installed and linked via brew.

Compiling fontconvert throws the following error.

```
make fontconvert
gcc -Wall -I/usr/local/include/freetype2 -I/usr/include/freetype2 -I/usr/include fontconvert.c -lfreetype -o fontconvert
fontconvert.c:22:10: fatal error: 'ft2build.h' file not found
#include <ft2build.h>
         ^~~~~~~~~~~~
1 error generated.
make: *** [fontconvert] Error 1
```

When i am adding homebrew's location of freetype there's this error:

```
make fontconvert
gcc -Wall -I/usr/local/include/freetype2 -I/usr/include/freetype2 -I/usr/include -I/opt/homebrew/include/freetype2 fontconvert.c -lfreetype -o fontconvert
ld: library 'freetype' not found
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [fontconvert] Error 1
```

Do you have any idea how to resolve the issue?

mitch-spec, 2024-01-18:
It compiles as soon as i am adding the following:
`CFLAGS = -Wall -I/usr/local/include/freetype2 -I/usr/include/freetype2 -I/usr/include **-L/opt/homebrew/opt/freetype/lib -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/opt/libpng/include/libpng16**`

mikeysklar, 2026-10-16:
Resolved with PR #463, which is now merged. Thanks for reporting this!

closed 2026-10-16 by mikeysklar