adafruit/Adafruit-GFX-Library #452: Natively support for degree symbol in FreeSans fonts

qwertz2021, 2024-06-16:
Hello,

is it possible to add/include the degree symbol in the FreeSans fonts in the future? Some users modyfied some font types on her own but there is no fully modified font packge with all font sizes and types on github.

https://github.com/ImpulseAdventure/GUIslice-Builder/issues/27 

If somebody has a modified conclumerat with degree symbo of FreeSans font feel it would be great if you could upload it here for.

Regards

iotricity, 2024-12-30:
I think it's the most wanted/requested symbol in all default fonts of Adafruit_GFX for years. Since the character table allows skipping of characters, why not add ASCII 176 (0xB0) or 248 (0xF8) by default?

mikeysklar, 2026-11-09:
Thanks for the request. The shipped FreeSans fonts stop at character 126, and changing them would break existing sketches.
You can make one with the degree sign in the browser: https://adafruit.github.io/Adafruit-GFX-Library/fontconvert/
Pick FreeSans.ttf, set last char to 176 (0xB0), and print it with `"25" "\xB0" "C"`.
Typing `°` in the Arduino IDE saves two UTF-8 bytes, so it draws the wrong glyph.

[image]

<details><summary>Test sketch</summary>

```cpp
// #452: a degree sign with a FreeSans font made by fontconvert,
// range 32 to 176 (0xB0 is the degree sign in Latin-1).
// Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing.
#include <Adafruit_GFX.h>
#include <Adafruit_HX8357.h>
#include "FreeSans12pt8b.h"

Adafruit_HX8357 tft(0, 15, -1); // CS, DC, RST

void setup() {
  tft.begin(8000000);
  tft.setRotation(3);
  tft.fillScreen(HX8357_BLACK);
  tft.setFont(&FreeSans12pt8b);
  tft.setTextColor(HX8357_WHITE);
  tft.setTextSize(2);
  tft.setCursor(20, 80);
  tft.print("25" "\xB0" "C"); // split so \xB0C isn't read as one hex escape
  tft.setCursor(20, 170);
  tft.print("77" "\xB0" "F");
  tft.setTextSize(1);
  tft.setTextColor(HX8357_YELLOW);
  tft.setCursor(20, 250);
  tft.print("FreeSans12pt8b, chars 32 to 176");
}

void loop() {}
```
</details>

Closing this one.

closed 2026-11-09 by mikeysklar