adafruit/Adafruit-GFX-Library #453: Prebuild fontconvert

ikhoeyZX, 2024-06-23:
i did tried build fontconvert and still no luck (freetype2 also not detected)
there is any precomplied binary (fontconvert,exe) that can used instead build by self (since building took too much my data plan and very slow because my country speed (only 25KB/s) for downloading tools 

```
$ make
gcc -Wall -I c:/mingw/include/freetype2 fontconvert.c -lfreetype -o fontconvert
/usr/lib/gcc/x86_64-pc-msys/13.2.0/../../../../x86_64-pc-msys/bin/ld: cannot find -lfreetype: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:8: fontconvert] Error 1
```

BillyDonahue, 2024-12-18:
I know you're in an unusual situation. I've privately rewritten fontconvert in Python, which might be useful for you.

https://github.com/adafruit/Adafruit-GFX-Library/pull/371

So you don't have to actually compile any C code to run a fontconvert.

mikeysklar, 2026-11-08:
There's now a browser version of fontconvert, nothing to build or install: https://adafruit.github.io/Adafruit-GFX-Library/fontconvert/
It gives the same output as the command-line tool. Closing this one.

closed 2026-11-08 by mikeysklar