adafruit/Adafruit-GFX-Library #462: Adafruit_GFX_32.h: No such file or directory

figsgksss, 2025-01-19:
c:\Users\alex\Documents\Arduino\libraries\Adafruit_ILI9488/Adafruit_ILI9488.h:22:10: fatal error: ../Adafruit_GFX_32/Adafruit_GFX_32.h: No such file or directory
 #include "../Adafruit_GFX_32/Adafruit_GFX_32.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

Please help

Armadillocurry, 2025-02-04:
i had the exact same error
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/library.properties#L10
this means u need to install the Adafruit BusIO library
it worked for me

mikeysklar, 2026-10-21:
Adafruit_GFX_32.h is not part of this library.

The include points to an Adafruit_ILI9488 folder on the reporter's machine.
That library is not this one, so please ask its maintainer.

Installing BusIO will not fix this one.

Closing this one.

closed 2026-10-21 by mikeysklar