adafruit/Adafruit-GFX-Library #468: GFXcanvas16 with drawRGBBitmap() causes core panic

FrankS71, 2025-04-02:
Adafruit TFT FeatherWing - 3.5" 480x320 and Adafruit ESP32 Feather V2 - 8MB Flash + 2 MB PSRAM

When using GFXcanvas16 canvas(480, 320) and drawRGBBitmap(0,0,canvas.getBuffer(),480,320) it causes core panic. canvas(240, 150) is working on this screen but only fills a small portion of the screen.

On smaller screens (Feather v2 reverse tft) it works fine.

mikeysklar, 2026-10-30:
Reproduced on Feather ESP32 V2 + HX8357.
A 480x320 canvas needs 300 KB of PSRAM.
With PSRAM disabled, getBuffer() is NULL and drawRGBBitmap panics.

Set Tools > PSRAM to Enabled and it draws in 162 ms.
Checking `canvas.getBuffer()` before drawing also catches it.

PR #533 stops the panic on a NULL buffer.

I used Claude Code to test this.

FrankS71, 2026-10-30:
Disabled PSRAM was not my issue at the time. It was probably Adafruit-GFX-Library psram handling or allocating
My solution was this (simplified code):

```cpp
#ifndef BOARD_HAS_PSRAM
  #error "The display requires PSRAM. Enable PSRAM in the Arduino IDE Tools menu."
#endif

#include <Adafruit_GFX.h>
#include "Adafruit_HX8357.h"
#define TFT_CS     15  // Chip select pin
#define TFT_DC     33  // Data/Command pin
#define TFT_RST    -1  // Reset pin
#define HX8357_PURPLE 0x8000
#define CANVAS_WIDTH  480
#define CANVAS_HEIGHT 320

Adafruit_HX8357 display = Adafruit_HX8357(TFT_CS, TFT_DC, TFT_RST);

// USE PSRAM for GFXcanvas16 if possible
GFXcanvas16* canvas = nullptr;

void setup() {

  if (!psramInit()) {
    Serial.println(" - PSRAM failed to initialize");
  } else {
    Serial.println(" - PSRAM initialized");
  }
  
  //Before
  Serial.printf("\n %-12s%-12s%-12s\n", "PSRAM Size", "PSRAM Used", "PSRAM Free");
  Serial.printf(" %10d%12d%12d\n", ESP.getPsramSize(), ESP.getPsramSize() - ESP.getFreePsram(), ESP.getFreePsram());
  
  canvas = new GFXcanvas16(CANVAS_WIDTH, CANVAS_HEIGHT);

  if (!canvas || !canvas->getBuffer()) {
    Serial.println("- ERROR: Could not allocate display canvas");
    delete canvas;
    canvas = nullptr;
  } else {
    Serial.println(" - Display canvas allocated");
    display.begin();
    display.setRotation(1);
    // Do graphic stuff
    //
   // And the draw to screen
    display.drawRGBBitmap(0, 0, canvas->getBuffer(), CANVAS_WIDTH, CANVAS_HEIGHT);
  }

  //After
  Serial.printf("\n %-12s%-12s%-12s\n", "PSRAM Size", "PSRAM Used", "PSRAM Free");
  Serial.printf(" %10d%12d%12d\n", ESP.getPsramSize(), ESP.getPsramSize() - ESP.getFreePsram(), ESP.getFreePsram());

}

void loop() {
  // put your main code here, to run repeatedly:
}

```

mikeysklar, 2026-10-30:
Thanks, you're right, PSRAM was not the cause.
A global canvas is built before setup(), before PSRAM joins the heap.
Retested on Feather ESP32 V2 with PSRAM enabled:

| canvas | result |
|---|---|
| global `GFXcanvas16 canvas(480, 320)` | 0x0, buffer NULL |
| `new GFXcanvas16(480, 320)` in setup() | 480x320, buffer in PSRAM |

So your fix is the right one.
PR #533 stops the panic when the buffer is NULL.

I used Claude Code to test this.

closed 2026-10-30 by mikeysklar