adafruit/Adafruit-GFX-Library #469: SPI peripheral mapping not working on some RP2040 bases boards

caternuson, 2025-04-09:
This forum poster was trying to get a [Feather RP2040 CAN](https://www.adafruit.com/product/5724) working with a [TFT FeatherWing](https://www.adafruit.com/product/3651):
https://forums.adafruit.com/viewtopic.php?t=217465

Dealing with ensuring the various CS pins were being handled properly worked to a certain point, however, it was not possible to get a full working example using both CAN and TFT at the same time. I tried to work up an example and ran into a similar issue. Drilling into things, it seems like an issue with some conditional mapping being done to set up the SPI peripheral being used. There are numerous instances of this code:
https://github.com/adafruit/Adafruit-GFX-Library/blob/87e15509a9e16892e60947bc4231027882edbd34/Adafruit_SPITFT.cpp#L1034-L1035

However, it seems the philhower core maps `spi0` and `spi1` to the  RP2040's SPI0 and SPI1 peripherals. The Feather RP2040 CAN's default SPI pins use the SPI1 peripheral:

[image]

also:
https://github.com/earlephilhower/arduino-pico/blob/e05dd50d626d2871ab988d8e72aa7c12e18e603c/variants/adafruit_feather_can/pins_arduino.h#L32-L37

So it seems like that tertiary logic ends up mapping to `spi0` instead of `spi1` as needed.

Doing a quick hack and changing all instances of:
```cpp
spi_inst_t *pi_spi = hwspi._spi == &SPI ? spi0 : spi1;
```
to:
```cpp
spi_inst_t *pi_spi = spi1;
```
resolves the issue.

So the problem is known, but at this point unclear what a good code fix would be.

**Example test sketch:**
```cpp
/*
 * TFTFeatherWing + Feather RP2040 CAN Example
 */
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <Adafruit_MCP2515.h>

#define MCP_CS PIN_CAN_CS   // CAN
#define STMPE_CS 6          // touch
#define TFT_CS   9          // TFT
#define TFT_DC   10
#define SD_CS    5          // SD
   
#define CAN_BAUDRATE (250000)

// CAN
Adafruit_MCP2515 mcp(MCP_CS);

// TFT
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

void setup() {
  Serial.begin(115200);
  while(!Serial) delay(10);

  Serial.println("TX Feather RP2040 CAN + TFT FeatherWing test!");

  // force disable all SPI things
  pinMode(MCP_CS, OUTPUT); digitalWrite(MCP_CS, HIGH);
  pinMode(STMPE_CS, OUTPUT); digitalWrite(STMPE_CS, HIGH);
  pinMode(SD_CS, OUTPUT); digitalWrite(SD_CS, HIGH);
  pinMode(TFT_CS, OUTPUT); digitalWrite(9, HIGH); 

  Serial.println("CAN init");
  if (!mcp.begin(CAN_BAUDRATE)) {
    Serial.println("Error initializing MCP2515.");
    while(1) delay(10);
  }
  Serial.println("MCP2515 chip found");

  Serial.println("TFT init");
  tft.begin();
  uint8_t x = tft.readcommand8(ILI9341_RDMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDPIXFMT);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDIMGFMT);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDSELFDIAG);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX);

  // fill color pattern
  tft.fillScreen(ILI9341_RED);
  delay(5000);
  tft.fillScreen(ILI9341_GREEN);
  delay(5000);
  tft.fillScreen(ILI9341_BLUE);
  delay(5000);

  Serial.println("START DONE");
}

void loop() {
  // send packet: id is 11 bits, packet can contain up to 8 bytes of data
  Serial.print("Sending packet ... ");

  mcp.beginPacket(0x12);
  mcp.write('h');
  mcp.write('e');
  mcp.write('l');
  mcp.write('l');
  mcp.write('o');
  mcp.endPacket();

  Serial.println("done");

  delay(1000);

  // send extended packet: id is 29 bits, packet can contain up to 8 bytes of data
  Serial.print("Sending extended packet ... ");

  mcp.beginExtendedPacket(0xabcdef);
  mcp.write('w');
  mcp.write('o');
  mcp.write('r');
  mcp.write('l');
  mcp.write('d');
  mcp.endPacket();

  Serial.println("done");

  delay(1000);
}
```

**CURRENT BEHAVIOR:**
[image]

**BEHAVIOR WITH HACK FIX:**
[image]

caternuson, 2025-04-09:
Here's the related PR that brought in these changes:
https://github.com/adafruit/Adafruit-GFX-Library/pull/360

caternuson, 2025-04-09:
grepping in the [`variants` subfolder of the RP2040 BSP](https://github.com/earlephilhower/arduino-pico/tree/master/variants) for boards that map `spi1` to `__SPI0_DEVICE`:
```
$ grep -r __SPI0_DEVICE | grep spi1
adafruit_feather_thinkink/pins_arduino.h:#define __SPI0_DEVICE  spi1
adafruit_feather_prop_maker/pins_arduino.h:#define __SPI0_DEVICE               spi1
amken_revelop_es/pins_arduino.h:#define __SPI0_DEVICE     spi1
adafruit_feather_dvi/pins_arduino.h:#define __SPI0_DEVICE  spi1
adafruit_metro_rp2350/pins_arduino.h:#define __SPI0_DEVICE   spi1
adafruit_feather_scorpio/pins_arduino.h:#define __SPI0_DEVICE  spi1
adafruit_feather_rfm/pins_arduino.h:#define __SPI0_DEVICE  spi1
adafruit_feather_can/pins_arduino.h:#define __SPI0_DEVICE     spi1
adafruit_feather_usb_host/pins_arduino.h:#define __SPI0_DEVICE   spi1
silicognition_rp2040_shim/pins_arduino.h:#define __SPI0_DEVICE  spi1
sparkfun_iotnode_lorawanrp2350/pins_arduino.h:#define __SPI0_DEVICE spi1
adafruit_feather_adalogger/pins_arduino.h:#define __SPI0_DEVICE  spi1
```

This issue would affect all of those boards.

caternuson, 2025-04-09:
This works for the CAN Feather, but not a universal fix since not all boards have these defines: 
```cpp
    spi_inst_t *pi_spi = hwspi._spi == &SPI ? __SPI0_DEVICE : __SPI1_DEVICE;
```
The BSP has defaults, but in .cpp file:
https://github.com/earlephilhower/arduino-pico/blob/e05dd50d626d2871ab988d8e72aa7c12e18e603c/libraries/SPI/src/SPI.cpp#L419-L424

ladyada, 2025-05-04:
did this get resolved with the PR?

mikeysklar, 2026-10-22:
Thanks for the report, @caternuson.
PR #470 picks the SPI block from the core's device macros.
I checked that a plain Feather RP2040 still draws correctly after it.

[image]

| Board | Display | Default SPI | fillScreen | Result |
|---|---|---|---|---|
| Feather RP2040 | 3.5" HX8357 FeatherWing | spi0 | 202 ms | red, green, blue, box |

I don't have a spi1 board here (CAN, RFM, Adalogger), so that case is untested.
Leaving this open until someone confirms on one of those.

<details><summary>Test sketch</summary>

```cpp
// #469: GC9A01A on the board's default SPI with an RP2040 Feather.
// On CAN/RFM/Adalogger/ThinkInk Feathers, SPI is spi1; on the plain Feather it is spi0.
// Pass: solid red, green, blue, then a white box on black, every loop.
// Build with -DUSE_HX8357 for the 3.5" HX8357 FeatherWing (CS 9, DC 10, no reset).
#include <Adafruit_GFX.h>
#include <SPI.h>

#ifdef USE_HX8357
#include <Adafruit_HX8357.h>
Adafruit_HX8357 tft(9, 10, -1);
#define RED HX8357_RED
#define GREEN HX8357_GREEN
#define BLUE HX8357_BLUE
#define BLACK HX8357_BLACK
#define WHITE HX8357_WHITE
#else
#include <Adafruit_GC9A01A.h>
Adafruit_GC9A01A tft(9, 10, 11);
#define RED GC9A01A_RED
#define GREEN GC9A01A_GREEN
#define BLUE GC9A01A_BLUE
#define BLACK GC9A01A_BLACK
#define WHITE GC9A01A_WHITE
#endif

void step(const char *name, uint16_t color) {
  uint32_t t = micros();
  tft.fillScreen(color);
  Serial.printf("%s %lu us\n", name, (unsigned long)(micros() - t));
  delay(1500);
}

void setup() {
  Serial.begin(115200);
  uint32_t t0 = millis();
  while (!Serial && millis() - t0 < 3000) delay(10);
  Serial.printf("BEGIN SPI on %s\n", __SPI0_DEVICE == spi1 ? "spi1" : "spi0");
  tft.begin();
}

void loop() {
  step("red", RED);
  step("green", GREEN);
  step("blue", BLUE);
  tft.fillScreen(BLACK);
  tft.fillRect(60, 60, 120, 120, WHITE);
  Serial.println("box");
  delay(1500);
  Serial.println("END");
}
```

Built with `-DUSE_HX8357`, GFX 1.12.6, arduino-pico on `adafruit_feather`.

</details>

Tested with help from Claude Code; I checked the hardware runs myself.

mikeysklar, 2026-10-23:
I tested the spi1 case on a Feather RP2040 Adalogger, whose default `SPI` is spi1.
Before #470 it hangs on the first fill; with 1.13.0 it draws.

[image]

| GFX | Serial | Screen |
|---|---|---|
| 1.12.0 (before #470) | nothing | stays white after power up |
| 1.13.0 | `BEGIN SPI on spi1`, 202 ms per fill | red, green, blue, box, loops |

Resolved with PR #470, which is now merged. Thanks for reporting this!

<details><summary>Test sketch</summary>

```cpp
// #469: GC9A01A on the board's default SPI with an RP2040 Feather.
// On CAN/RFM/Adalogger/ThinkInk Feathers, SPI is spi1; on the plain Feather it is spi0.
// Pass: solid red, green, blue, then a white box on black, every loop.
// Build with -DUSE_HX8357 for the 3.5" HX8357 FeatherWing (CS 9, DC 10, no reset).
#include <Adafruit_GFX.h>
#include <SPI.h>

#ifdef USE_HX8357
#include <Adafruit_HX8357.h>
Adafruit_HX8357 tft(9, 10, -1);
#define RED HX8357_RED
#define GREEN HX8357_GREEN
#define BLUE HX8357_BLUE
#define BLACK HX8357_BLACK
#define WHITE HX8357_WHITE
#else
#include <Adafruit_GC9A01A.h>
Adafruit_GC9A01A tft(9, 10, 11);
#define RED GC9A01A_RED
#define GREEN GC9A01A_GREEN
#define BLUE GC9A01A_BLUE
#define BLACK GC9A01A_BLACK
#define WHITE GC9A01A_WHITE
#endif

void step(const char *name, uint16_t color) {
  uint32_t t = micros();
  tft.fillScreen(color);
  Serial.printf("%s %lu us\n", name, (unsigned long)(micros() - t));
  delay(1500);
}

void setup() {
  Serial.begin(115200);
  uint32_t t0 = millis();
  while (!Serial && millis() - t0 < 3000) delay(10);
  Serial.printf("BEGIN SPI on %s\n", __SPI0_DEVICE == spi1 ? "spi1" : "spi0");
  tft.begin();
}

void loop() {
  step("red", RED);
  step("green", GREEN);
  step("blue", BLUE);
  tft.fillScreen(BLACK);
  tft.fillRect(60, 60, 120, 120, WHITE);
  Serial.println("box");
  delay(1500);
  Serial.println("END");
}
```

Built with `-DUSE_HX8357` for `rp2040:rp2040:adafruit_feather_adalogger`, arduino-pico 5.6.0, 3.5" HX8357 FeatherWing.

</details>

Tested with help from Claude Code; I checked the hardware runs myself.

closed 2026-10-23 by mikeysklar